rodrigomelo9/verifying-foss-hdl-synthesizers #1: Take a look at SymbiFlow/fpga-tool-perf

mithro, 2020-03-24:
Firstly, thanks for doing work to improve Yosys compatibility with existing Xilinx code bases. The work is greatly appreciated! I'm very interested in supporting any work which makes Yosys a more viable synthesis frontend.

You might find the project at https://github.com/SymbiFlow/fpga-tool-perf interesting. The project aims to compare performance of Vivado (and maybe in the future ISE) and the open source tools on important designs. It's more focused on "whole designs" and performance rather than compatibility but could still be interesting to you.

We have recently ported the project t use [@olofk's EDAlize project](https://github.com/olofk/edalize) for interacting with the FPGA toolchains.

There are also a simple design doc around this project at https://docs.google.com/document/d/16L50pyS3RjYStvRKRoWyVac7NoZKyu45fpPQXJqeKYo/edit

BTW  You might also find the following things interesting;
 - Collaboration between Google, Uni of Toronto, Xilinx and SymbioticEDA around FPGA interchange formats which is been worked on at https://interchange.vtr.tools/
 - SystemVerilog linter and code formatter - https://github.com/google/verible
 - SystemVerilog parser + preprocessor - https://github.com/alainmarcel/Surelog
 - SystemVerilog Test Suite - https://symbiflow.github.io/sv-tests/ && https://github.com/SymbiFlow/sv-tests

rodrigomelo9, 2020-03-25:
Thanks Tim, you are welcome.

I found `fpga-tool-perf` some days ago (suggested by GitHub in the "Explore repositories" section). Again, I can see the similarities between Edalize and PyFPGA (I am in contact with Olof and I will evaluate a meld or cooperation). Yes, the repository is interesting to me, to see whole projects working :-), thanks.

The collaboration, where also Xilinx is involved, is very interesting. Is Xilinx agreeing to implement such formats in the future as part of its tools? Are there more information?

And about the SystemVerilog repositories... I am a VHDL boy, who knows how to use Verilog (most of my "real world" projects are using VHDL, is the language that I select when I face a new design). It is the reason because I didn't start before with Yosys. I never used SystemVerilog for a project, I only take one or two small courses where I saw a part of the syntaxis, no more than that.

eine, 2020-04-06:
*ping* @tgingold, as he might be interested on the FPGA interchange formats from GHDL's perspective. Also [UHDM](https://github.com/alainmarcel/UHDM) feels conceptually very close to [ghdl-systemc-fosdem16](https://github.com/ghdl/ghdl-systemc-fosdem16).

rodrigomelo9, 2020-05-04:
@mithro Now I know about the existence of this project and where to see if needed. Regards.

closed 2020-05-04 by rodrigomelo9