rodrigomelo9/verifying-foss-hdl-synthesizers #8: Usage of docker images from ghdl/docker

eine, 2020-04-06:
It seems that you are using, or suggesting to use, docker images that are maintained/released from ghdl/docker, in order to avoid building ghdl and ghdl-yosys-plugin. Precisely image `ghdl/synth:beta` is used in https://github.com/rodrigomelo9/verifying-foss-hdl-synthesizers/blob/master/helpers/ghdl.sh

I just wanted to let you know that image `ghdl/synth:beta` does include a fully usable yosys (apart from ghdl-yosys-plugin and ghdl). Hence, you can update the README not to require installing yosys locally. Furthermore, image `ghdl/synth:yosys` exists (which `ghdl/synth:beta` is based on). This includes yosys only, without ghdl or the plugin.

Other related images are available with `nextpnr`, `symbiyosys`, etc. See: https://github.com/ghdl/docker#-cache-5-jobs--max-4--11-images-weekly. Usage examples are available at [antonblanchard/ghdl-yosys-blink](https://github.com/antonblanchard/ghdl-yosys-blink), https://github.com/ghdl/ghdl-yosys-plugin/blob/master/examples/docker.mk and https://github.com/eine/vhdl-cfg/blob/master/GHDLSynth.py. However, this issue is not about suggesting you to drop your scripts in favour of any of the existing solution. Instead, I just want to let you know about this resources, so that you can hopefully find insteresting ideas/solutions to apply to your own projects.

Last, but not least, for any visiting reader in this issue, I feel that most of `ghdl/synth` docker images do not really correspond to ghdl's namespace. @mithro is open to hosting them in Symbiflow, so `ghdl/synth:yosys`, `ghdl/synth:nextpnr*`, `ghdl/synth:symbiyosys`, etc. might be moved to `symbiflow/*` in the near future (as soon as we have time to decouple them from ghdl/docker). See context in ghdl/ghdl-yosys-plugin#95.

rodrigomelo9, 2020-04-07:
Hi @eine, thanks for the data. The fact is, that I have Yosys compiled and updated in my PC (which is useful to check a solved reported bug), but not GHDL (I added support for it recently). I will unify it in some way.

rodrigomelo9, 2020-05-04:
Hi @eine, I am using the ghdl/synth:beta image as the base of my own Dockerfile (which adds PyFPGA) and using it from CI in this repository. I mentioned it in the new README, but let me know if you feel that some more need to be told. Thanks.

closed 2020-05-04 by rodrigomelo9

eine, 2020-05-04:
@rodrigomelo9, more than enough. There is absolutely no attribution requirement when using images from ghdl/docker. Take the resources in that repo as public domain. Strictly, it would be anti-copyright, should there be a person/company that owned the authorship. That is not the case. Nonetheless, thanks a lot for mentioning the source!

BTW, I have a couple of issues to reply to you. I've been quite busy and I'll do when I have some time to cleanup first.

commit bd300820 (2020-05-01): Updated README.md

Closes #8 #10.