rodrigomelo9/verifying-foss-hdl-synthesizers #10: Use of conda to get latest version of tools

rodrigomelo9, 2020-04-07:
Hi @mithro, I read your document, but I need more info about that point :P I will check how to use Conda, and I need an environment which supports:
* Yosys, GHDL and the ghdl-yosys-plugin, iverilog, verilator.
* Python3 and edalize.
* ISE, Vivado and Quartus.

Recommendation/suggestions? How you solve the support for ISE/Vivado? (which means virtual in your document?). Thanks in advance for links or material that you could provide.

mithro, 2020-04-07:
> Hi @mithro, I read your document, but I need more info about that point :P I will check how to use Conda, and I need an environment which supports:

FYI -- If you don't want to support **all** of Linux, Mac and Windows then conda is not really the right choice. If you just want to support Linux, then something like Docker is probably a better / simpler option.

I have [some conda notes in a Google Doc here](https://docs.google.com/document/d/1AaL460Uf_8_9v8EIy4pAZYTFOFXbv5O4RRU7Jv4t_Z4/edit) which explain various things. 

Conda uses `environment.yml` files for specifying environments, these can include the traditional Python `requirements.txt` files. See https://github.com/SymbiFlow/sv-tests/blob/master/.github/travis/install.sh and https://github.com/SymbiFlow/sv-tests/blob/master/conf/environment.yml

> * Yosys, GHDL and the ghdl-yosys-plugin, iverilog, verilator.

Yosys / iverilog / verilator should all be up to date (but I would like to get faster at pushing new packages).

GHDL and ghdl-yosys-plugin are still a WIP but should be added in the near future.

> * Python3 and edalize.

This is pretty easy.

Example script -- https://github.com/litex-hub/litex-conda-packages/blob/master/conda-get.sh

> * ISE, Vivado and Quartus.

These are not ready yet but should be in a couple of weeks.

> Recommendation/suggestions? How you solve the support for ISE/Vivado? (which means virtual in your document?). Thanks in advance for links or material that you could provide.

FYI - EDDA is a current work in progress and rapidly evolving.

rodrigomelo9, 2020-04-07:
No, I am not interested in Windows or Mac here (I assume that how the tools work in these OS is tested somewhere :P). So, I probably will use docker images described at #8, adding what needed.

mithro, 2020-04-07:
@rodrigomelo9 - You could have a play with conda and see what you think, but I'm guessing docker will be a better solution for you.

rodrigomelo9, 2020-05-04:
Hi @mithro, I finally used Decker. Thanks.

closed 2020-05-04 by rodrigomelo9