rodrigomelo9/verifying-foss-hdl-synthesizers #9: Alternative version of fpga-hdl2bit using edalize

rodrigomelo9, 2020-04-07:
I will implement a version of `fpga-hdl2bit` based on edalize, to check similarities and differences. @mithro

mithro, 2020-04-07:
FYI -- @kgugala and @olofk can probably help here.

rodrigomelo9, 2020-04-07:
I will use `fpga-tool-perf` as base example, to implement the minimum needed here. Maybe when implemented they can verify if correctly used to take advantages of its features.

For what is done in this repository, what I need is something like:
```
fpga-hdl2bit [--outdir OUTDIR] [--tool TOOL] [-f DEP1] [-f DEP2] [--run TASK] TOPFILE
```
Where:
* `OUTDIR` is where files are generated. It implies needed Tcls, Makefiles or bash scripts, and also the output product of the vendor tools. It is a desired option, to avoid to populate with hundreds of files.
* `TOOL` is the tool to use and can be: `yosys` (only synthesis), `ise`, `vivado`, `quartus`, `yosys-ise` (Yosys with ise for P&R), `yosys-vivado` (Yosys with vivado for P&R). Also, `ghdl` (only for synthesis`), `ghdl-yosys` (only for synthesis) and, `ghdl-yosys-ise` and `ghdl-yosys-vivado' would be useful, to support all the used tools in this repo (https://gitlab.com/rodrigomelo9/pyfpga/-/issues/34).
* `DEPx` are modules (files) involved in the design, which are not the top-level.
* `TOPFILE` is mandatory, and the top-level of the design. A file is provided and the top-level name is automatically detected. If more than one top-level is detected, the last one is taken.

The fpga-hdl2bit implementation based on PyFPGA is [here](https://gitlab.com/rodrigomelo9/pyfpga/-/blob/master/fpga/helpers/hdl2bit.py).

olofk, 2020-04-08:
Edalize supports ise, vivado, ghdl and quartus already
There is a PR for yosys-vivado in Edalize which needs a bit of cleaning up but after that it's ready to go. It would be awesome to add ghdl synthesis support. I guess the easiest way would be to extend the yosys class to accept VHDL files too. If you're interested I'm happy to accept patches for that.

But if we ignore the ghdl synthesis stuff for a while, I think the easiest way to support edalize in fpga-2hdlbit would be to parse the command-line, build an [EDAM](https://edalize.readthedocs.io/en/latest/edam/api.html) object from that info and pass it on to Edalize

rodrigomelo9, 2020-04-08:
Hi @olofk, thanks. What about to specify only to perform synthesis or synthesis + P&R (without bitstream generation). As far I understood (reading the README, and with a quick view of the source code) it can't be specified. I am wrong? I will implement the alternative version to learn edalize (and probably I could contribute things).

My entry-point API, fpga.project, solves some high-level things, such as generate and run all into an specified directory, time measurement, top-level autodetection when a file is provided (instead of provide the name by hand, and moreover, the last top-level in the file is selected, which is useful here because there are Xilinx examples with multiple modules in a single file). Probably, fpga.project could use edalize instead of my EDA interface. I must further investigation :P but I saw that for example, when passing parameters, you must specify if `vlogparam` or `generic`. Is it needed because are there some tools which need these differences? With the tools that I actually support (ise, vivado, quartus, libero-soc and yosys) is not needed. I am only asking to know because I saw these differences.

Regards

olofk, 2020-04-08:
The icestorm, quartus, trellis and vivado backends (but not diamond and ise) have options to set `pnr : none` in the tool_options section to only perform synthesis.

Good point about vlogparam and generic. I had to check, and all the backends that support both types actually specify them in the same way. I hadn't noticed that before :) I originally wanted to separate them since there are some differences to when they are applied. Verilog parameters must be fixed during compilation (for simulations) while generics can be set when you run the simulation model, more like plusargs. In reality though, none of the Edalize backends take advantage of this.

rodrigomelo9, 2020-05-04:
I played a little with edalize. I need things such as run only synthesis (which is supported) and synthesis + P&R (as far I understood, it is not possible, edalize run P&R and bitstream generation together, let me know if I am wrong). Moreover, it seems that the tools are called multiple times, which is a big problem in case of Vivado, which takes at least 10 seconds, but sometimes 40, to start (again, maybe I am wrong, let me know). There are a lot of examples analyzed in this repository and I prefer not to waste time invoking several times the tools. Anyway, I am using `fpga-hdl2bit` in this repository, the thing is, If I use `PyFPGA` or `edalize` under-the-hood, which I think that if needed, could be treated in [other place](https://gitlab.com/rodrigomelo9/pyfpga/-/issues/26).

Please, @olofk let me know if I am wrong in something that I said about edalize and @mithro, any thought will be welcome.